VHDL/Compliance-Tests #7: Using Python3 in Vunit

Paebbels, 2019-10-16:
Is it possible to get rid of old Python2 syntax and use more up-to-date Python 3 with e.g. `PathLib` instead of nasty `join(("foo", "bar"))`?  
Does VUnit methods support PathLib?

https://github.com/VHDL/Compliance-Tests/blob/3eae5fb1063faffb54204c852b483f0fa0be2a9f/run.py#L22

Python3 PathLib:

    root = Path("/tests")
    sources = root / "src"

1138-4EB, 2019-10-16:
According to http://vunit.github.io/about.html#python: *VUnit support for Python 2.7 will end in 1 Jan 2020 when it reaches end of life*.

eine, 2020-02-02:
FTR, examples in branch `master` of VUnit are already modified to use `Path`. As soon as the next release is published, we can modify the scripts in this repo.

Paebbels, 2020-02-02:
That sounds great.

It's interesting how many projects switch from supporting a dinosaur to python >=3.5. This transitions seems to happen quite fast all over the Internet. The deprecation of Python 2.7 seemed to be a signal every one waited for, but one one wanted to touch it themselves ...

eine, 2023-02-26:
This was done in https://github.com/VHDL/Compliance-Tests/commit/9278be966e96574dd03d2327e538b048a89e1eee.

closed 2023-02-26 by eine