VHDL/Compliance-Tests #11: LCS2019: automatic issue generation, MWE update and testing tools

eine, 2020-02-02:
Branch [LCS2019](https://github.com/VHDL/Compliance-Tests/tree/LCS2019/issues) contains a helper script to allow projects to automatically create one GitHub issue for each Language Change Specifications (LCS) in the recently published VHDL 2019. The list of LCS is defined in a YAML file, and they can optionally include a MWE.

Currently, very few LCS contain an MWE. Hence, help is wanted to write examples and/or to copy existing ones from http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/VHDL2017.

Furthermore, [issue-runner](https://github.com/eine/issue-runner) allows MWEs defined in GitHub issues to be automatically tested in CI workflows. It is suggested to make MWEs compatible with it. See https://eine.github.io/issue-runner/#usage.

bpadalino, 2023-02-26:
I don't see an LCS2019 branch anymore, so I assume it has been pulled in at some point.  I am currently trying to get this done with my [`skeleton-2019`](https://github.com/bpadalino/Compliance-Tests/tree/skeleton-2019) branch and PR draft #13.

Once that is merged in, do you think this issue has been resolved?

eine, 2023-02-26:
Branch LCD2019 contained https://github.com/VHDL/Compliance-Tests/tree/main/issues, which was merged into main. We can close this issue. We will continue in #13 and other PRs.

closed 2023-02-26 by eine