VHDL/Compliance-Tests #16: GitHub Step Summary

umarcor, 2023-02-26:
Some months ago, GitHub introduced https://github.blog/2022-05-09-supercharging-github-actions-with-job-summaries/. Essentially, `$GITHUB_STEP_SUMMARY` points to a "file" where you can write markdown code. Then, it's shown in the workflow web view, beloew the graph. See, for instance: https://github.com/hdl/containers/actions/runs/4080049545.

In this repo, we currently need to open the workflow, expand the step and scroll to the botton in order to see the results. That's because we exit green unconditionally, so the exit status of CI is not meaningful by itself. It would be nice if we printed something in the summary.

@LarsAsplund what would be the best approach to get the summary from VUnit? Shall we do it in the `run.py` or generate an XUnit report and the process it in a different script?

LarsAsplund, 2023-02-27:
I was thinking about a JSON+XUnit export and have a VUnit attribute on tests referencing relevant LRM section. The output would be a table with test case name + pass/fail + LRM link. Now that we have two simulators that we can use online we should have a column for each. Just like in that SystemVerilog example. We should also generate a HTML report for those running offline with their own simulator. Pandoc on the Markdown maybe. My current client uses GitHub and there I use this technique to create a test to spec mapping which the CI pushes to the GitHub Wiki. I didn't know about this summary feature so let's give it a try. Will you? I better focus on the VUnit support for NVC.

umarcor, 2023-03-07:
We are talking about different use cases. The GitHub Step Summary is useful for someone watching a CI run through the web. So clicking on any of the runs listed in https://github.com/VHDL/Compliance-Tests/actions.

Generating JSON, XUnit or HTML cannot be done there. That needs an additional step, which might be a) downloading an artifact and/or b) pushing it somewhere (gh-pages, wiki, anywhere else).

In [ghdl/extended-tests](https://github.com/ghdl/extended-tests), we use `pytest pytest-reporter pytest-reporter-html1` and we upload the HTML report to gh-pages (see [ghdl.github.io/extended-tests](https://ghdl.github.io/extended-tests/)).
Moreover, in GHDL we pregenerate the content of a subdir and then include it in the doc: https://github.com/ghdl/ghdl/blob/master/doc/conf.py#L114-L115. Hence, we might convert VUnit's XUnit output to HTML and include it in the documentation.

PR #30 (2023-03-07): run: use GITHUB_STEP_SUMMARY when running within GitHub Actions
Close #16.

This PR prints the list of failed tests to the GITHUB_STEP_SUMMARY (https://github.blog/2022-05-09-supercharging-github-actions-with-job-summaries/), so that we don't need to open the logs in order to see that.

Results are obtained through the `post_func` feature of VUnit.
closed 2023-03-07 by LarsAsplund